Pheneeny/CuraPlugin-ScalableExtraPrime #3: settings don't show even when plugin is activated.

MajorSlug, 2018-03-27:
Settings fields are well set visible:
[image]

But don't show up.

[image]

To access them, I need to use the "search..." bar.

[image]

rooiejoris, 2018-04-16:
same here, could this be in relation with the json file...?!

cheers / joris

Pheneeny, 2018-09-22:
I don't know the root cause of the issue, but I know that if you collapse the "Material" category completely and re-expand it, the settings will show up properly from then on.

I apologize for the late response, I did not have my notifications configured in Github correctly and was never notified of your issue.

MajorSlug, 2018-09-22:
Thanks for the tip.

closed 2020-11-27 by Pheneeny