Pheneeny/CuraPlugin-ScalableExtraPrime #4: Combing only?

samtaylor3d, 2018-05-02:
Hey fantastic job. One thing I'd love to see world be an option to select only coming. I have no problems when I retract but combing gives me dry starts. Great work and thanks again. Not really an issue but I'm a newbie and couldn't find a message or comment option

Pheneeny, 2018-09-22:
Sorry for the late response, I did not have my notifications configured in Github correctly and was never notified of your issue.

I think that is a good idea for an enhancement, I will try to implement it in the next release I do.

Pheneeny, 2020-11-27:
Added ability to separately toggle if extra prime should be added for travels with retractions and travels without retractions.

closed 2020-11-27 by Pheneeny