Pheneeny/CuraPlugin-ScalableExtraPrime #6: Retraction in End G-Code also modified by plugin.

MozeLabs, 2018-05-04:
[image]
[image]
[image]

As picture and title said. It add the retraction value (E-2 and E-3 in my setting) with the last offset value (116.58898 in this case).
Maybe it could be set to stop modifying code after G91 command (for my setting, i don't know how other printers settings could be).

Garyr14, 2018-06-30:
I was having the same problem and finally figured out what was causing it!

agibson2, 2018-09-08:
I had the same problem.  I just removed retraction from my end gcode to fix it.  My end gcode moves the nozzle up and to home position so any oozing happens not over the print... well there is a tiny bit that it does leave on the print as it lifts the head but that is minor compared to what it was doing.