Pheneeny/CuraPlugin-ScalableExtraPrime #11: Can't install in Cura 3.6

rlesac, 2018-11-16:
It's not listed in the marketplace and drag and drop just produces an error.
Please update!

mcer12, 2018-12-01:
I suggest you to revert to 3.4.1 as I did. 3.6 doesnt introduce anything super useful so I dont mind. And lets hope it gets approved soon.

IllyStorm, 2019-02-15:
Any updates on availability in Cura 3.6? Finding myself having to guess my "extra prime" setting depending on if i'm doing a model with long travels or not - this plugin seems like a good solution

agibson2, 2019-04-04:
It doesn't work with 4.0 either.  This plugin was very useful as it fixes a lot of situations where underextrusion would happen on medium to long travels.  I hope that it does eventually get compatibility fixes for newer Cura versions.  Maybe we can get together and put some $$$ toward it for someone to get it working with newer Cura versions.

stirlsilver, 2019-04-28:
Another vote to get this working for Cura 4.0!

zombalo, 2019-05-10:
Hello everyone, i just figured out how to make this awesome plugin work with cura 4.0 and 3.6 (tested only on 4.0)

Just change content fo plugin.json file to this

To get plugin installed, go to curaInstallFolder/plugins
create path 'ScalableExtraPrime/ScalableExtraPrime' and put inside last folder all repository files

Then change plugin.json file content to following

`{
  "name": "Scalable Extra Prime",
  "author": "Pheneeny",
  "author_email": "pheneeny@gmail.com",
  "website": "https://github.com/Pheneeny/CuraPlugin-ScalableExtraPrime",
  "version": "1.0.1",
  "description": "Adjusts the extra prime amount after a travel or retraction based on the distance traveled",
  "api": 4,
  "supported_sdk_versions": ["5.0.0", "6.0.0"]
}
`
Also when you set checkbox for activating this plugin, nested settings doesn't appear in options list, you can see them only searching options names from readme.md, i has no idea how to fix this issue so this is workaround

Elfish, 2019-10-10:
I'm also heavily interested in a fix for 4.x

HankHillPropane, 2020-02-10:
For Cura 4.4.X I had to change the API value in the json file to version 7, and not 4 as stated above.  Works great now.

hnolsoe, 2020-07-06:
I Cura 4.6 I got this feature visible by altering the json file:

    "api": 5,
    "supported_sdk_versions": ["5.0.0", "6.0.0", "7.0.0"]

However a notch in the "Enable Scalable Extra Prime" option does NOT make the 4 subcategories visible (surely a bug) unless you go up top and search for "prime".

agibson2, 2020-07-09:
The workaround for the 4 sub-settings not being visible is already mentioned by Elfish 2 posts above yours or are you saying that didn't work for you?

hnolsoe, 2020-07-09:
Oh ... collapsing the Material section and reopening it makes those 4 sub-settings visible permanently as mentioned above - just missed those comments the first time around.